Geeetech3D/Smartto-Eclipse #4: Make sure Marlin 2 supports the hardware

superjamie, 2018-06-08:
Marlin 1 was an 8-bit Arduino only project. Marlin 2 changes to using a HAL and will also support 32-bit boards such as LPC1768 and STM32, see: https://github.com/MarlinFirmware/Marlin/issues/7076

To add new board support, usually Marlin project members to buy hardware themselves then reverse engineer the functionality themselves from datasheets and experimentation. The board manufacturers/vendors do not help out. This is not ideal.

Geeetech should work with the Marlin project to ensure the Smartto board is supported in Marlin 2 on release. Use your technical experience on the STM32 platform to complete the Marlin STM32 HAL, and ensure the `pins.h` for the Smartto board is contributed to Marlin by Geeetech employees. This would be a good friendly open source contribution.

The ideal outcome is that all users and developers of the Marlin project just have to `#define MOTHERBOARD` for the Smartto, and board support "just works" with Marlin firmware. The Marlin project should not have to perform much reverse engineering or figuring out implementation things that the board vendor could help with.

thatjoshguy, 2018-06-09:
I support this idea.  While I applaud Geeetech for releasing Smartto as Open Source, having the option of running Marlin on the hardware would be a huge bonus and would make Geeetech look good in the community's eyes. Marlin has a much larger user base and therefore there are more developers contributing code.

Geeetech3D, 2018-06-11:
Thanks all, as you said it's good in the commucnity's eyes, so we will also run the Marlin on our STM32 board later. we also do some work on the smartto more than marlin, for example 1) Receive and transmit G-code commands through ESP8266 Wi-Fi module  2)Support 3.2 inch full color touch screen;3)Machine parameters can be configured without compilation, including all motor parameters and printing volume

closed 2018-06-28 by Geeetech3D

superjamie, 2018-06-28:
Normally you would only close an Issue when the request has been completed.

I do not see any code in Marlin's `bugfix-2.0.x` branch which adds support for this board, so the work requested on this Issue has not been done.

This Issue should not be Closed.

prahjister, 2018-06-30:
Yes please open this feature request back open. If possible a wiring diagram with pinout would be excellent.

kaktusus, 2018-06-30:
The Circuit Diagram has been available since the establishment of this repository ...
Look in this branch:
circuit_diagram