Geeetech3D/Smartto-Eclipse #5: [BUG] Pausing the Print/Filament Runout Cools Down Bed

stampede-arm, 2018-06-12:
When the printer is paused or the filament runout occurs, the machine turns off both heating elements. The bed cools down, and if you don't catch it right away, releasing the part from the bed and making it impossible to resume. Ideally, the bed stays on.

Geeetech3D, 2018-06-13:
that's a good suggestion, we will change it in next version that the bed will keep heating when printer is paused.

crncrussell, 2018-06-19:
It would be good if it could stay on for maybe 30 mins, then timeout and turn off as a safety measure.

stampede-arm, 2018-06-20:
Ah! Yes, a timer of some sorts would be good. Same energy and stay safe
while still giving the user a chance.

Suggy50, 2018-07-03:
This issue still here in the latest update

SonOfSardaar, 2018-11-21:
Issue is still there

kaktusus, 2018-11-21:
[ When you are tormented by nasty pauses during printing.](url=https://www.geeetech.com/forum/viewtopic.php?f=77&t=61729)