Geeetech3D/Smartto-Eclipse #8: MacOS Applications for Firmware Upgrade

redangel1984, 2018-06-22:
I have a Mac-only environment, and I would like to have the latest firmware on my A30. Is it possible to upload/uograde the Firmware from a Mac computer. Is it possible to do this? Will this be possible in the near future?

Geeetech3D, 2018-06-25:
Hi redangel1984, I'm sorry for that there is no Mac version of our tools now. and will be add this to Mac.
we will fix this bug and add more other gcode to smartto like Marlin firmware. Thanks redangel!

Geeetech3D, 2018-06-28:
@redangel1984  We have uploaded a firmware tool for Mac already, you can try to upgrade your machine now. If you have more suggestions for the tool or other softwares, please contact us.
As for Motor. we will push new tool for Mac in few days too. Please focus on it.

redangel1984, 2018-06-28:
[image]

redangel1984, 2018-06-28:
What now?

Geeetech3D, 2018-06-30:
@redangel1984  Em... it seems like your MacOS needs to be upgraded. Build environment of the tools was base on OS X 10.13.3. We will upload tools supported OS X 10+ in a few days or you can upgrade your system for this one.

redangel1984, 2018-06-30:
I can’t upgrade my system. My Macs hardware I use doesn’t support MacOS 10.11+
If you upgrade the tool to support MacOS 10.11 that would be great. Thanks in advance!