Geeetech3D/Smartto-Eclipse #9: Not a issue, a request for a new feature, Set step/mm from display

marcofanuntza, 2018-06-22:
Hello, this post is not for a issue problem, is a request for a new function, could be very nice  set the steps-per-unit for one or more axes directly from display, like Marlin Firmware do.....

i have a Geeetech A30, thanks!

Geeetech3D, 2018-06-25:
Hi marcofanuntza, thanks for your good sugesstion! we will add more settings on the smartto_motor_tool.exe, not in A30 display now because of that it will be complicated for some beginners

thatjoshguy, 2018-06-28:
Using the tool requires a Windows PC, and definately not easy for beginners. These settings as well as jerk and acceleration, max speed, etc.  need to be configurable on the touch screen without the need for a PC.

closed 2018-07-25 by Geeetech3D