Geeetech3D/Smartto-Eclipse #11: HARDWARE circuit_diagram/Rostock301/

kaktusus, 2018-06-25:
Due to a design error with regard to the comparator, I suggest not to mount the following parts:
JP1
Q4, Q1, IC5
And the related passive elements.

It is often the case that an accidental short-circuit of the fan leads to damage to the mosfet transistor (Q4). This one gets punctures and supplies the fans with 24V DC supply voltage.
Also applies to a board built on µC AVR.
I had such a case myself, and I read about similar things in the forum.

Kind regards.
kakrus

Geeetech3D, 2018-06-28:
@kaktusus  that's good for us,because we are redesigning this hardware.

kaktusus, 2018-06-28:
I would be happy to take part in the design of a new controller.

closed 2018-07-25 by Geeetech3D