Geeetech3D/Smartto-Eclipse #12: Suggestion: Play beep sound

kaktusus, 2018-06-26:
Hi
1. The Gcode can be used to generate audio signals.
At present, sending a code to the controller:
M300 P10 S100
M30 P50 S200
Parameters
    Snnn frequency in Hz
    Pnnn duration in milliseconds

will generate one short sound, not a longer modulated one.
Separate queuing of audio commands could be considered.

2. I suggest you generate short beeps after using the encoder button, or each time you rotate the encoder.
There may be an additional option in the menu to disable these sounds.

Can this be taken into account in development plans?

kaktus

Geeetech3D, 2018-06-28:
Thanks @kaktusus  ! we will add this M300 gcode

redsleep, 2018-06-28:
Hello there:

Thank you so much for your suggestion.
In the later development, I will consider the command control and button control of the sound.
We hope that you can provide more valuable suggestions for us later, and we will seriously consider them.

I hope you have a nice day!

Jone

kaktusus, 2018-06-28:
Jone, you can count on me to do that.

closed 2018-07-25 by Geeetech3D