Geeetech3D/Smartto-Eclipse #14: Suggestion: New Features - Monitor Printer Life Cycle

kaktusus, 2018-07-04:
Hi
Let Geeetech printers be the first with this functionality.

I propose to create a functionality that will make it possible:
- global counting of the time the printer is on, and current indication, as it was done now,
- global work time counting (when the printer is printing), and the current duration of the most recent print,

- filament consumption calculation in the last printing process for each extruder and separately for the total of
- global filament consumption counting for individual extruders, and additionally together.

These features will allow you to estimate component usage and monitor the printer's life. On this basis, many other useful conclusions can be drawn.

Geeetech3D, 2018-07-05:
Good advice. Life cycle monitoring is an essential feature in 3d printer. In fact, some of parameters you mentioned had been used in our PC client. But there still a lot of error in time/filament consumption calculation. So we would not put that into the printer until we can provide accurate digits.

Geeetech3D, 2018-07-05:
@kaktusus

Geeetech3D, 2018-07-05:
add the filament counting is a great idea! but not sure how prescise if it calculate from gcode file.
or another way, add a count sensor

kaktusus, 2018-07-05:
In my opinion, it is possible to monitor the operation of the feeder (stepper motor of the extruder). After all, the number of steps needed to move out a 1mm filament is known.

You can also count the number of filament serving from Gcode commands.
Here, there may be discrepancies due to not performed operations when the work is interrupted and the counter can already be increased or decreased.

It seems to me that it will be easier and less burdensome to monitor the work of stepper motors and on this basis count the shift of the filament on + and -.

I wouldn't build additional gadgets to count the filament when we have such a computing power and background. I mean this powerful microcontroller to supervise the printer.

But yes, your new temperature measuring container can be perfected with this function. After all, not everyone has this great printer controller (GTM32Pro). This will be a great feature for users of printers with another controller board, who will purchase this container.

counting the time when the printer is powered on is not a complicated function.
One pin can be connected to +24V via optocoupler. Add a capacitor to the microcontroller and examine the pulse loss on the selected optocoupler pins. Count the time at the same time.
When the power supply disappears, proceed immediately to write down the value of the time counted.
An algorithm for distributed recording of EEPROM memory cells needs to be developed.
The problem may be to develop a method for counting the printing time.
It would be necessary to cunningly invent triggering and stopping the timer.
At the moment I don't have an idea how easy it is to do it.

kaktusus, 2018-07-05:
It is certainly easy to count the working time of each heater.
Collecting this information is also very interesting.

kaktusus, 2018-07-06:
Monitoring of the next value.
You may be familiar with TI chips, INA2XX series, e.g. INA226 [Current/Power Monitor](url=http://www.ti.com/power-management/protection-monitoring-hot-swap/i2c-pmbus/products.html#p929=I2C&p726max=28;80). ;-D
A great ready-made solution.

kaktusus, 2018-07-06:
To measure the working time (printing) of the printer, I suggest to prepare a simple mechanism.

Create a 10 second interval. _(10s counter)_
Each time you receive a Gcode G0 or G1 command, the counter _(10s)_ is reset.
The printer's working time counter is active during this time.
When the 10s counter is overflowing because the G0 or G1 command no longer comes in, the printer's operating time measurement is finished.

The proposed solution is simple and should not interfere with the operation of the printer.

It can be problematic to detect pauses when printing the same object.