Geeetech3D/Smartto-Eclipse #16: Z too quick after homing -> it looses steps

redangel1984, 2018-07-05:
I tried both the new firmwares: 1.37.59 and 1.37.60.
and I cannot home reliable. After levelling and homing, when it recovers to Z offset value. the stepper motors are too quick, and the Z-axis binds and loses steps. It is so bad that i can't print anything with it.
If i restore older firmware everything works fine.
if you need more info about this issue, feel free to contact me

kaktusus, 2018-07-05:
I will ask you out of curiosity
what model of printer you have 
How many stepper motors do the Z-axis have?
and how many stepstik modules do you use to control the Z axis?

redangel1984, 2018-07-05:
A30
2 Stepper motors/lead screws 
1 Stepper Driver

thatjoshguy, 2018-07-06:
test: if you send the printer gcode command "G1 Z0 F800" from home does it result in the same behavior?  This appears to be called at the end of the G28 homing routine.  Maybe replace Z0 with the positive value of your Z offset to recreate the behavior if that doesn't work.

If so try lowering F800 to something lower until you find something that works that isn't too slow.  Post up your best value here

redangel1984, 2018-07-06:
I will try it tonight. But isn’t 800mm/min a little too much for the Z axis? 13.3mm/s
Was that the same for the older 1.35.56 firmware too, because on the older one everything homed well. I can’t modify anything on the firmware because I have a Mac and cannot compile the modified firmware. I can only install already compiled ones.

Geeetech3D, 2018-07-06:
When the motor is too fast after you encounter the updated firmware, you can try to restore the factory settings. Today updated the latest firmware V1.37.61 of the motherboard, you can try it.

redangel1984, 2018-07-06:
Where can I try this latest 1.37.61 firmware?
As I said I cannot compile the source code( cannot install the tools for it)
Could you make a folder on github with the compiled firmware so that we can try the latest realeases without needing to compile it.

Or maybe you could add it to:

geeetech.com/firmware/

Geeetech3D, 2018-07-06:
@redangel1984 we put that into /firmwares/test

Geeetech3D, 2018-07-06:
https://github.com/Geeetech3D/Smartto/tree/release/firmwares/IAR

redangel1984, 2018-07-06:
I looked and searched but don’t find any “test” folder or “ *.bin “ file. Did I miss something?
Thanks for your patience and effort!!!

redangel1984, 2018-07-06:
Found it!!! Thanks 🙏

closed 2018-07-25 by Geeetech3D