Geeetech3D/Smartto-Eclipse #20: Request: change naming of firmware files

thatjoshguy, 2018-07-06:
Currently the firmware for the A30 is distributed in two files, an "S" and and "M" file.  One is for the screen, and one is for the main board... 

For English speaking users one would assume the "S" file is for the Screen, and the "M" file is for the Main board... but this is not the case!  The "M" file is for the screen, and vice versa.

When distributing future firmware releases, could they contain the words "SCREEN" and "MAIN" in their file names so there is no confusion?

Geeetech3D, 2018-07-07:
A reasonable request. At the beginning, we set 'M' for Master device and 'S' for Slave device. But it seems that making users confused. We're going to make an internal discussion about it.

kaktusus, 2018-07-07:
Each description is good provided it is explained in the readme file, which is located in the main branch of the project.

thatjoshguy, 2018-07-07:
Master and Slave does make sense in that context. If it was noted in the readme that would be good.

closed 2018-07-25 by Geeetech3D