Geeetech3D/Smartto-Eclipse #21: Discrepency in Eclipse project, IAR project, and firmware releases

thatjoshguy, 2018-07-06:
What is the proper procedure for contributing code to the SMARTTO project? Currently there are two branches, each with two different projects and code bases (Eclipse and IAR).

Changes I submitted that were merged into the Eclipse project had errors that were corrected in the IAR code updated by (I am assuming) Geetech.  

Why are we maintaining two separate versions of essentially the same code?  Isn't this going to cause issues further down the line?  Obviously someone noticed the error in my pull request, why was it accepted?  It should have been rejected so I could fix it and then merged into master.  Why was it only fixed in the IAR version?

I want to support this project and help SMARTTO grow, but I am afraid that will be difficult unless it is made clear how to go about this and we move to a single code base/project. Eclipse is free and easy to use so I think it makes a good choice to promote it as the primary code base

thatjoshguy, 2018-07-06:
On a related note: when compiling from Eclipse I get a single .bin file but normally the firmware releases include two files, one for main board and one for screen.  Is there a different way I should compile for testing, or does the Eclipse project not include the screen firmware?

Geeetech3D, 2018-07-07:
Thanks, Josh. We are working on it .

closed 2018-07-25 by Geeetech3D