Geeetech3D/Smartto-Eclipse #24: Endstop logic - NO or NC

redangel1984, 2018-07-26:
Hi,
I want to install an Optical Z Endstop, and discovered that the logic of the optical sensors are reversed (NC) is there a possibility to change the logic of the sensor through Gcode or through the configuration tool? So that if :
Circuit closed - untriggered
Circuit open   - triggered

Geeetech3D, 2018-07-30:
Hello there
Currently, gcode command modification is not supported.
You can modify the configuration file configuratio_GTM32_A30.h by yourself.
Modify the macro definition below：
*****************************************************************************************
#define X_MIN_ENDSTOP_LEVEL    HIGH  //LOW  //HIGH
#define X_MAX_ENDSTOP_LEVEL    HIGH
#define Y_MIN_ENDSTOP_LEVEL     HIGH
#define Y_MAX_ENDSTOP_LEVEL    HIGH
#define Z_MIN_ENDSTOP_LEVEL     HIGH
#define Z_MAX_ENDSTOP_LEVEL    HIGH

#define X_MIN_ENDSTOP_STATUS     CLOSE    //OPEN  CLOSE
#define X_MAX_ENDSTOP_STATUS     CLOSE
#define Y_MIN_ENDSTOP_STATUS     CLOSE
#define Y_MAX_ENDSTOP_STATUS    CLOSE
#define Z_MIN_ENDSTOP_STATUS     CLOSE
#define Z_MAX_ENDSTOP_STATUS    CLOSE
*******************************************************************************

redangel1984, 2018-07-30:
Thanks for your answer. I know where to modify and what.
But the problem is I cannot compile the Firmware because i dont have Windows, and that is why I suggested that you could create a custom Gcode to change that value for example:

**M217 X P0 S0** ; example for changing X min LOW,  (P0-min P1-max) , S0-LOW,  S1-HIGH
**M217 Z P1 S1** ; example for changing Z max to HIGH,  (P0-min P1-max) , S0-LOW,  S1-HIGH
etc.
Or to add it to the Motor Tool, so users can change it without needing to recompile the firmware.
And you could rename the Motor Tool to: **"Smartto Configuration Tool"** and include all the values which you can change

Suggy50, 2018-07-30:
Came here looking for the same solution but redangel1984 as beaten me to the question.
I have tried to do this through Eclipse but not having any look importing the whole project.
As said above it would be nice to have the option in Motor Tool

Geeetech3D, 2018-08-01:
hi
Thank you very much for your suggestion. In the next version we will add a command line to modify the configuration.

Suggy50, 2018-08-01:
Thanks you
Much appreciated, do you have a time scale on the release date?

Suggy50, 2018-10-13:
Any news on the new release, still wanting to add optical censor