Geeetech3D/Smartto-Eclipse #25: Changing controller fan strength by GCODE

wlky, 2018-08-13:
Hey,
Not sure if there already is a way to change the controller fan strength but I didn't see anything in the gcode file.

I think it would be something that could easily be added to the start and end gcode in order to increase the cooling of the controller during the print and decrease it after.

alexwoo1900, 2018-08-14:
@wlky 
Gcode is generated by your slicer engine. Slicer has corresponding parameters to set it up.
In gcode file, M106 for changing fan strength and M107 for shutting the fan down

wlky, 2018-08-14:
I'm talking about the controller box fan speed, not the nozzle fan speed

kaktusus, 2018-08-14:
Hi 
Probably not all the sockets support such functionality.

alexwoo1900, 2018-08-14:
@wlky 
Same. Depend on your parameter "P"
If you set M106 P0 Sxxx, it will change the nozzle fan speed.
If you set M106 P2 Sxxx, it will change the controller box fan speed.

wlky, 2018-08-14:
@alexwoo1900 thanks! That's what I was looking for.

closed 2018-08-14 by wlky